periph/conn #13: Race when host.Init() is called

eriksejr, 2021-08-19:
**Describe the bug**
I've been running a project which uses the MCP23XXX driver on ARM64 (RPI4) under the race detector to debug a few things. I've fixed all the races I have run into except this one which I hit almost every time the program starts up when host.Init() is called:

```
==================
WARNING: DATA RACE
Write at 0x00c0001179e0 by main goroutine:
  runtime.mapaccess2_faststr()
      /usr/local/go/src/runtime/map_faststr.go:107 +0x48c
  periph.io/x/conn/v3/driver/driverreg.(*stage).loadParallel()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg_parallel.go:101 +0xb4
  periph.io/x/conn/v3/driver/driverreg.initImpl()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg_parallel.go:55 +0x32c
  periph.io/x/conn/v3/driver/driverreg.Init()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg.go:59 +0xd0
  periph.io/x/host/v3.Init()
      /home/pi/go/pkg/mod/periph.io/x/host/v3@v3.7.0/host.go:18 +0x30
  main.(*wingleBerryDevice).periphInit()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:137 +0x3c
  main.newDevice()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:124 +0xc68
  main.main()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:323 +0xd8

Previous read at 0x00c0001179e0 by goroutine 21:
  runtime.mapaccess1_faststr()
      /usr/local/go/src/runtime/map_faststr.go:12 +0x45c
  periph.io/x/conn/v3/driver/driverreg.(*stage).loadParallel.func1()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg_parallel.go:76 +0x274

Goroutine 21 (running) created at:
  periph.io/x/conn/v3/driver/driverreg.(*stage).loadParallel()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg_parallel.go:69 +0x80
  periph.io/x/conn/v3/driver/driverreg.initImpl()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg_parallel.go:55 +0x32c
  periph.io/x/conn/v3/driver/driverreg.Init()
      /home/pi/go/pkg/mod/periph.io/x/conn/v3@v3.6.8/driver/driverreg/driverreg.go:59 +0xd0
  periph.io/x/host/v3.Init()
      /home/pi/go/pkg/mod/periph.io/x/host/v3@v3.7.0/host.go:18 +0x30
  main.(*wingleBerryDevice).periphInit()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:137 +0x3c
  main.newDevice()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:124 +0xc68
  main.main()
      /home/pi/gostuff/src/wbcmdd/wbcmdd.go:323 +0xd8
==================
```

Line 137 of wbcmdd.go calls host.Init():

```
135 func (wb *wingleBerryDevice) periphInit() error {
136         // Initialize periph.io
137         if _, err := host.Init(); err != nil {
138                 return err
139         } else if ......
```
I am also using periph.io for GPIO on the Raspberry Pi 4. When I remove the hardware that uses the MCP23017 chip but call host.Init() to still initialize periph.io for the GPIO I do not see this race. It seems like it may be related to this specific driver.

Note that I am not actually having any issues with the driver or the software, only that I am running into this race being reported when run under the race detector.

**Expected behavior**
No race is reported

**Platform (please complete the following information):**
OS: Raspberry Pi OS 64-bit
Board: Raspberry Pi 4 8GB RAM version

maruel, 2021-08-20:
https://github.com/periph/conn/blob/v3.6.8/driver/driverreg/driverreg_parallel.go#L76 reads loaded in a separate routine while line 101 writes to it. I guess it's really faster than other platforms because I never detected it elsewhere.

maruel, 2021-08-20:
Thanks a lot for the bug report. It made is easy to find the bug.

https://github.com/periph/conn/tree/sync should resolve this. I have to log off now, will investigate tomorrow to get it in and make a new release.

maruel, 2021-08-20:
I fixed in 1d85d07b4615168a7cc82950da52504f8317b2fe bug forgot to refer to the bug. I have an on-going PR that I'll try to get committed then I'll tag another release. In the meantime, explicitly fetching this commit should get you going, e.g.:

`go get periph.io/x/conn/v3@1d85d07b4615168a7cc82950da52504f8317b2fe`

but it's a matter of a few days.

closed 2021-08-20 by maruel