periph/conn #14: gpioreg in v3 fails to register and list All() GPIOs on IMX8 aarch64

romatou18, 2021-09-01:
**Describe the bug**
Trying to list GPIOs using 
```go
import v3gpioreg "periph.io/x/conn/v3/gpio/gpioreg" 
```

gpioreg.All() fails to list the gpios on the board len() == 0

Similarly any attempt go get hold of a pint by name fails just the same i.e.
```go 
p := v1gpioreg.ByName("GPIO8")
```

When using the old library lets call it v1 using instead:
```go 
import v1gpioreg "periph.io/x/periph/conn/gpio/gpioreg"
```

This works perfectly. 

**To Reproduce**

1. Run program
```go
package main

import (
	"fmt"
	"log"

	"periph.io/x/conn/v3/driver/driverreg"
	v3gpioreg "periph.io/x/conn/v3/gpio/gpioreg"
	v1gpio "periph.io/x/periph/conn/gpio"
	v1gpioreg "periph.io/x/periph/conn/gpio/gpioreg"
	v1host "periph.io/x/periph/host"
)

func getGPIOSystemIndex(port, index int) int {
	name := ((((port) - 1) * 32) + ((index) & 31))
	alternative := (port-1)*32 + index

	if name != alternative {
		log.Fatalf("name and alternative are different ! %d %d\n", name, alternative)
	}
	return name
}

func main() {

	// V3 gpioreg based does not work no gpio found
	if _, err := driverreg.Init(); err != nil {
		log.Fatal(err)
	}
	fmt.Printf("gpioreg len = %d, ALL(): %+v \n", len(v3gpioreg.All()), v3gpioreg.All())

	if len(v3gpioreg.All()) < 1 {
		fmt.Println("v3gpioregFailed to read GPIOs ! using v3 conn.")
	}

	// old repo version, using pinreg works first hand ! easy !
	if _, err := v1host.Init(); err != nil {
		fmt.Println(err)
	}

	fmt.Printf("v3gpioreg len = %d, ALL(): %+v \n", len(v3gpioreg.All()), v3gpioreg.All())

	if len(v3gpioreg.All()) < 1 {
		fmt.Println("Failed to read GPIOs ! using v3 conn gpioreg.")
	}

	fmt.Printf("v1gpioreg len = %d, ALL(): %+v \n", len(v1gpioreg.All()), v1gpioreg.All())

	if len(v1gpioreg.All()) < 1 {
		fmt.Println("Failed to read GPIOs ! using v1 conn gpioreg.")
	}

	gpioLinuxIndex := getGPIOSystemIndex(1, 8) // GPIO1_IO08 => port 1 gpio 08 VAR-DT8MCustomBoard_Datasheet.pdf
	log.Println("gpio sys index = ", gpioLinuxIndex)
	gpioSysName := fmt.Sprintf("GPIO%d", gpioLinuxIndex)
	log.Println("gpio name = ", gpioSysName)

	// get the specified GPIO pin
	p := v1gpioreg.ByName(gpioSysName)
	if p == nil {
		log.Fatal("Failed to find " + gpioSysName)
	}

	// Switch LEd on
	if err := p.Out(v1gpio.High); err != nil {
		log.Fatal(err)
	}

}

```
2. Run it.
3. See error

**Expected behavior**
```bash
root@imx8mp-var-dart:~# SmartHubGPIO_aarch64
gpioreg len = 0, ALL(): [] 
v3gpioregFailed to read GPIOs ! using v3 conn.
v3gpioreg len = 0, ALL(): [] 
Failed to read GPIOs ! using v3 conn gpioreg.
v1gpioreg len = 160, ALL(): [GPIO0 GPIO1 GPIO2 GPIO3 GPIO4 GPIO5 GPIO6 GPIO7 GPIO8 GPIO9 GPIO10 GPIO11 GPIO12 GPIO13 GPIO14 GPIO15 GPIO16 GPIO17 GPIO18 GPIO19 GPIO20 GPIO21 GPIO22 GPIO23 GPIO24 GPIO25 GPIO26 GPIO27 GPIO28 GPIO29 GPIO30 GPIO31 GPIO32 GPIO33 GPIO34 GPIO35 GPIO36 GPIO37 GPIO38 GPIO39 GPIO40 GPIO41 GPIO42 GPIO43 GPIO44 GPIO45 GPIO46 GPIO47 GPIO48 GPIO49 GPIO50 GPIO51 GPIO52 GPIO53 GPIO54 GPIO55 GPIO56 GPIO57 GPIO58 GPIO59 GPIO60 GPIO61 GPIO62 GPIO63 GPIO64 GPIO65 GPIO66 GPIO67 GPIO68 GPIO69 GPIO70 GPIO71 GPIO72 GPIO73 GPIO74 GPIO75 GPIO76 GPIO77 GPIO78 GPIO79 GPIO80 GPIO81 GPIO82 GPIO83 GPIO84 GPIO85 GPIO86 GPIO87 GPIO88 GPIO89 GPIO90 GPIO91 GPIO92 GPIO93 GPIO94 GPIO95 GPIO96 GPIO97 GPIO98 GPIO99 GPIO100 GPIO101 GPIO102 GPIO103 GPIO104 GPIO105 GPIO106 GPIO107 GPIO108 GPIO109 GPIO110 GPIO111 GPIO112 GPIO113 GPIO114 GPIO115 GPIO116 GPIO117 GPIO118 GPIO119 GPIO120 GPIO121 GPIO122 GPIO123 GPIO124 GPIO125 GPIO126 GPIO127 GPIO128 GPIO129 GPIO130 GPIO131 GPIO132 GPIO133 GPIO134 GPIO135 GPIO136 GPIO137 GPIO138 GPIO139 GPIO140 GPIO141 GPIO142 GPIO143 GPIO144 GPIO145 GPIO146 GPIO147 GPIO148 GPIO149 GPIO150 GPIO151 GPIO152 GPIO153 GPIO154 GPIO155 GPIO156 GPIO157 GPIO158 GPIO159] 
2021/09/01 06:38:41 gpio sys index =  8
2021/09/01 06:38:41 gpio name =  GPIO8
```

**Platform (please complete the following information):**
 - OS: yocto zeus 5.4 kernel +  sumo 4.14 based on Variscite yocto demo distribution
 - Board : aarch64 Variscite IMX8 and IMX8 PLUS
 https://variwiki.com/index.php?title=DART-MX8M-PLUS
https://variwiki.com/index.php?title=DART-MX8M

**Additional context**
The distribution used are the standard downloadable demo yocto using on the demo board VAR-DT8Customboard v1.4
All in all this is a market standard IMX8 board. Fails with both IMX8 and IMX8 plus Socs on Zeus kernel 5.4 and Sumo kernel 4.14 versions of yocto.

maruel, 2021-09-01:
v1 and v3 are incompatible. Use one or the other. See https://periph.io/news/2020/a_new_start/ for more information.

closed 2021-09-01 by maruel

romatou18, 2021-09-02:
@maruel I am not sure that you've read my report, this is not about trying both libs together but trying v3 first and watching it fail, then as for investigating trying the older version and watching it succeeding.
If this is what it takes i ll create a new bug report for v3 and show just that: on exactly the same OS and platform, same config, it fails to list gpios while it should not. 

Please take the time to read the report and investigate it this time. 

Thanks and regards