periph/conn #21: no communication with gpio pins on a raspberry pi zero 2 w

V0r-T3x, 2022-05-27:
I'm trying to install the [P4wnP1 aloa](https://github.com/RoganDawes/P4wnP1_aloa) project on a **raspberry pi zero 2 w** and it running on kali linux 2022.2. Originally it's supposed to be installed on a _raspberry pi zero w_. A feature let the user read and write the gpio pins to trigger some actions.

We already have an issue open on the P4wnP1 git page: [here](https://github.com/RoganDawes/P4wnP1_aloa/issues/307).
I take note on my installation [here](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/P4wn1_aloa-rpi02w.md).

The original maker of this project give us a hint on potential problem. He pointed [this part of the code](https://github.com/RoganDawes/P4wnP1_aloa/blob/95de406b72cd1c66f987184a2f7455fcae337252/service/SubSysGpio.go#L62L70) based on periph.io.

maruel, 2022-05-27:
This code is using the old library periph.io/x/periph. It should be updated to use periph.io/x/host/v3 and periph.io/x/conn/v3.

closed 2022-05-27 by maruel

V0r-T3x, 2022-05-27:
> This code is using the old library periph.io/x/periph. It should be updated to use periph.io/x/host/v3 and periph.io/x/conn/v3.

Thanks for fast respond. I will try this tonight, I hope this will solve my problem.