periph/conn #40: WaitForEdge Fails on Rasperry Pi Kernel 6.6.16-v8+

gsexton, 2024-02-12:
**Describe the bug**
Using the specified kernel, a call go WaitForEdge immediately unblocks and returns.

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
        "log"
        "periph.io/x/conn/v3/driver/driverreg"
        "periph.io/x/conn/v3/gpio"
        "periph.io/x/conn/v3/gpio/gpioreg"
        "periph.io/x/host/v3"
)

func main() {
    _, err:=host.Init()
    if err!=nil {
        panic(err)
    }
    _, err = driverreg.Init()
    if err!=nil {
        panic(err)
    }
    pin := gpioreg.ByName("GPIO19")
    pin.In(gpio.PullUp, gpio.BothEdges)
    for  {
        edgeReceived := pin.WaitForEdge(-1)
        if edgeReceived {
            log.Print("Edge Received., State=", pin.Read())
        } else {
            log.Print("No Edge Received")
        }
    }
}

```
2. Run it.
3. Note the output continuously scrolls "No Edge Received".

**Expected behavior**
The program should have blocked until the button was pressed.

**Platform (please complete the following information):**
 - OS: Raspbian Bookworm
 - Board: Pi 4 B 4GB
 - Kernel: 6.6.16-v8+

**Additional context**
The program has been working for several years. After a kernel update, it stopped working. Experimentation shows that reading the state of the button does work.

I've searched around, and found some references that this might be related to deprecation of /sys/class/gpio?

FWIW, I tried python and RPi.GPIO and it sees the button press as well, but attempt to add event for edge detection fails.

maruel, 2024-02-16:
Dang, sorry to hear. I fear someone will have to do the migration. Since the driver is in host, not in conn, it's tracked at https://github.com/periph/host/issues/39.

gsexton, 2024-09-02:
Closing this because https://github.com/periph/host/pull/59 resolves it.

closed 2024-09-02 by gsexton