periph/conn #41: How to work with UART?

eleimt, 2024-12-02:
**Describe the bug**
No UART

**To Reproduce**
Steps to reproduce the behavior:
1. Run program (https://github.com/periph/conn/blob/main/uart/uartreg/example_test.go#L46)
```go
	// Make sure periph is initialized.
	// TODO: Use host.Init(). It is not used in this example to prevent circular
	// go package import.
	if _, err := driverreg.Init(); err != nil {
		log.Fatal(err)
	}

	// Enumerate all UART ports available and the corresponding pins.
	fmt.Print("UART ports available:\n")
	for _, ref := range uartreg.All() {
		fmt.Printf("- %s\n", ref.Name)
		if ref.Number != -1 {
			fmt.Printf("  %d\n", ref.Number)
		}
		if len(ref.Aliases) != 0 {
			fmt.Printf("  %s\n", strings.Join(ref.Aliases, " "))
		}

		b, err := ref.Open()
		if err != nil {
			fmt.Printf("  Failed to open: %v", err)
		}
		if p, ok := b.(uart.Pins); ok {
			fmt.Printf("  RX : %s", p.RX())
			fmt.Printf("  TX : %s", p.TX())
			fmt.Printf("  RTS: %s", p.RTS())
			fmt.Printf("  CTS: %s", p.CTS())
		}
		if err := b.Close(); err != nil {
			fmt.Printf("  Failed to close: %v", err)
		}
	}
```
2. Run it.
3. See only "UART ports available:"

**Expected behavior**
UART list

**Platform (please complete the following information):**
 - OS: Linux pi 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64
 - Board: Raspberry Pi 4

eleimt, 2024-12-03:
@mattetti

gsexton, 2024-12-09:
@eleimt The normal architecture would be for a "periph.io/host" package to be included, and that "host" would as part of it's init locate all of the UARTs and register them with "uartreg".

I did a grep of the periph.io/host package and find no files referencing uartreg()

I remembered wondering about this a while ago, and here's the thread on it:

https://github.com/periph/host/issues/18

eleimt, 2024-12-11:
> @eleimt The normal architecture would be for a "periph.io/host" package to be included, and that "host" would as part of it's init locate all of the UARTs and register them with "uartreg".
> 
> I did a grep of the periph.io/host package and find no files referencing uartreg()
> 
> I remembered wondering about this a while ago, and here's the thread on it:
> 
> [periph/host#18](https://github.com/periph/host/issues/18)

I not understand how this using. Give me example please. Thanks.

gsexton, 2025-01-05:
@eleimt 

Here's a basic approach using a different library. The serial library for periph.io/conn is not implemented.

```go
package main

import (
	"errors"
	"fmt"
	"log"
	"time"
	//    "periph.io/x/conn/v3/gpio"
	"periph.io/x/conn/v3/display"
	"periph.io/x/conn/v3/display/displaytest"
	"periph.io/x/devices/v3/serlcd"
	// "periph.io/x/devices/v3/mow"
	"go.bug.st/serial"
)

func main() {
	fmt.Println("serlcd")
	mode := &serial.Mode{
		BaudRate: 9600,
		Parity:   serial.NoParity,
		DataBits: 8,
		StopBits: serial.OneStopBit,
	}
	port, err := serial.Open("/dev/ttyUSB0", mode)
	if err != nil {
		log.Fatal(err)
	}
	time.Sleep(time.Second)
	mow := serlcd.NewSerLCD(port, 4, 20)
	time.Sleep(time.Second)
	defer mow.Halt()
	mow.Clear()
	// mow.Contrast(40)
	time.Sleep(time.Second)
	mow.Display(true)package main

import (
	"errors"
	"fmt"
	"log"
	"time"
	"periph.io/x/conn/v3/display"
	"periph.io/x/conn/v3/display/displaytest"
	"periph.io/x/devices/v3/serlcd"
	"go.bug.st/serial"
)

func main() {
	fmt.Println("serlcd")
	mode := &serial.Mode{
		BaudRate: 9600,
		Parity:   serial.NoParity,
		DataBits: 8,
		StopBits: serial.OneStopBit,
	}
	port, err := serial.Open("/dev/ttyUSB0", mode)
	if err != nil {
		log.Fatal(err)
	}
	time.Sleep(time.Second)
	mow := serlcd.NewSerLCD(port, 4, 20)
	time.Sleep(time.Second)
	defer mow.Halt()
	mow.Clear()
	// mow.Contrast(40)
	time.Sleep(time.Second)
	mow.Display(true)
	time.Sleep(time.Second)
	fmt.Println(mow.String())
	n, err := mow.WriteString("abcdefg")
	time.Sleep(time.Second)
	fmt.Println(mow.String())
	n, err := mow.WriteString("abcdefg")
}
```

closed 2025-01-13 by gsexton