periph/conn #48: Unable to read output pin state.

larsks, 2025-07-18:
**Describe the bug**

I am unable to successfully read the output state of a GPIO pin on a raspberry pi 3.

**To Reproduce**

Here's some sample code that reproduces the problem:

```
package main

import (
	"fmt"
	"log"
	"time"

	"periph.io/x/conn/v3/gpio"
	"periph.io/x/conn/v3/gpio/gpioreg"
	"periph.io/x/host/v3"
)

func main() {
	// Initialize periph
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}

	// Get GPIO pin 23
	pin := gpioreg.ByName("23")
	if pin == nil {
		log.Fatal("Pin 23 not found")
	}

	// Set pin as output
	if err := pin.Out(gpio.Low); err != nil {
		log.Fatal(err)
	}

	// Set pin to high
	if err := pin.Out(gpio.High); err != nil {
		log.Fatal(err)
	}

	// Read value
	v := pin.Read()
	fmt.Printf("Set to high, read: %v\n", v)
	if v != gpio.High {
		log.Fatal("Expected high, got low")
	}

	time.Sleep(1 * time.Second)

	// Set pin to low
	if err := pin.Out(gpio.Low); err != nil {
		log.Fatal(err)
	}

	// Debug: check pin function
	fmt.Printf("Pin function: %s\n", pin.Function())

	// Read value
	v = pin.Read()
	fmt.Printf("Set to low, read: %v\n", v)
	
	// Try a small delay before reading
	time.Sleep(10 * time.Millisecond)
	v2 := pin.Read()
	fmt.Printf("After delay, read: %v\n", v2)
	
	if v != gpio.Low {
		log.Fatal("Expected low, got high")
	}

	time.Sleep(1 * time.Second)

	// Set pin to high again
	if err := pin.Out(gpio.High); err != nil {
		log.Fatal(err)
	}

	// Read value
	v = pin.Read()
	fmt.Printf("Set to high, read: %v\n", v)
	if v != gpio.High {
		log.Fatal("Expected high, got low")
	}

	fmt.Println("GPIO test completed successfully")
}
```

This fails with:

```
Set to high, read: High
Pin function: Out/High
Set to low, read: High
After delay, read: High
2025/07/18 13:49:26 Expected low, got high
```

On the other hand, the following Python code using the `gpiod` module works as expected:

```
import time
import gpiod

chip = gpiod.Chip('0')
switch = chip.get_line(23)
switch.request(consumer="testing", type=gpiod.LINE_REQ_DIR_OUT)

switch.set_value(1)
v = switch.get_value()
assert v == 1
time.sleep(1)

switch.set_value(0)
v = switch.get_value()
assert v == 0
time.sleep(1)

switch.set_value(1)
v = switch.get_value()
assert v == 1
```

This makes me believe the issue is with `periph` rather than with the hardware.

**Expected behavior**

I expect `pin.Read()` to accurately return the state of an output pin.

**Platform (please complete the following information):**
  
- OS: Raspbian GNU/Linux 12 (bookworm)
- Board: Raspberry Pi 3 Model B Rev 1.2

marcuscraske, 2025-08-06:
Seeing a similar issue with RPI 5.

````
Linux xxx 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
````

larsks, 2025-08-08:
I worked around this by porting my code over to https://github.com/warthog618/go-gpiocdev.

gsexton, 2026-01-12:
@larsks The documentation for the  periph.io/x/conn/v3/gpio/PinIn interface Read() function states:

_Behavior is undefined if In() wasn't used before._

In your example code, you're not calling In(). Essentially, you're trying to use the same opened pin for input and output.

Referencing the linux kernel docs:

[GPIO_V2_GET_LINE_IOCTL](https://docs.kernel.org/userspace-api/gpio/gpio-v2-get-line-ioctl.html).

_The direction flags, GPIO_V2_LINE_FLAG_INPUT and GPIO_V2_LINE_FLAG_OUTPUT, cannot be combined._

[GPIO_V2_LINE_GET_VALUES_IOCTL](https://docs.kernel.org/userspace-api/gpio/gpio-v2-line-get-values-ioctl.html)

_The values of both input and output lines may be read._

_For output lines, the value returned is driver and configuration dependent and may be either the output buffer (the last requested value set) or the input buffer (the actual level of the line), and depending on the hardware and configuration these may differ._

So what you're trying to do is non-deterministic across devices, and quite possibly will have varying results across different kernel versions with the same device.

It feels like your request is coming from one of two places. Either you're setting the pin value and want a quick and easy way to know the current state without saving it in memory, or you're trying to do bi-directional communications using one pin.

The first one is pretty easy. Save the current pin state as a variable and reference it.

If you're trying to do bi-directional communications using a single pin, periph isn't going to work. Honestly, nothing will work reliably because of the lack of contract from the kernel about what value you'll actually get. The best work-around in this case would be to use one GPIO pin for write, and another GPIO pin configured with Pull==Float for read.

closed 2026-01-12 by gsexton